mukmalone/hello_robot #30: Solving "ModuleNotFoundError: No module named 'gudev'"

pitosalas, 2021-06-15:
If this is a duplicate, I apologize. I wrote this issue 10 minutes ago and it's not showing up. Either I forgot to press "submit" or for some reason you decided to delete it. 

I am traveling the same road you seem to be traveling, about 1 week behind. I am using Linorobot, on noetic which requires python3. It is on ubuntu 20.04 - which is not exactly your configuration but I am hopeful that your solutions will work for me. I've also perused many of your issues to see if I could find a solution. I am well on the way but not there yet. The problems are with lino_install/install.sh (which I see you had to and recorded some changes.) I think further that the problems are because of python3.

The specific step that I am asking about is:

`~/linorobot_ws$ rosrun lino_udev lino_udev.py
`
which for me is giving this error: 

```
[onboard:bullet]~/linorobot_ws$ rosrun lino_udev lino_udev.py
Traceback (most recent call last):
  File "/home/ubuntu/linorobot_ws/src/linorobot_src/lino_udev/scripts/lino_udev.py", line 3, in <module>
    import gudev
ModuleNotFoundError: No module named 'gudev'

```
I have tried a bunch of things that looked like they would solve it but they've not worked. I suspect I am somewhat dirtying up my environment with these experiments. 

Do you have any tip or suggestion? Thanks!

mukmalone, 2021-06-15:
Take a look here for the solution  I found
https://github.com/mukmalone/hello_robot/issues/15#issuecomment-830270283

pitosalas, 2021-06-16:
Thanks a lot! Looks highly promising, I appreciate it! However, I hit this error, I wonder if you have seen it?

checking for Python headers using /usr/bin/python-config --includes... ./configure: line 9653: /usr/bin/python-config: No such file or directory
failed, will try another way
  File "<string>", line 1
    import sys; print sys.prefix
                      ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import sys; print sys.exec_prefix
                      ^
SyntaxError: invalid syntax
checking for Python headers in  and ... -I/include/python3.8
checking whether those headers are sufficient... no
configure: error: install python-devel

-------> Here's the full log. 

[onboard:bullet]~/catkin_ws/python-gudev$ ./autogen.sh

**Warning**: I am going to run `configure' with no arguments.
If you wish to pass any to it, please specify them on the
`./autogen.sh' command line.

processing .
Running libtoolize...
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
Running aclocal -I m4  ...
Running automake --gnu  ...
configure.ac:5: warning: AM_INIT_AUTOMAKE: two- and three-arguments forms are deprecated.  For more info, see:
configure.ac:5: https://www.gnu.org/software/automake/manual/automake.html#Modernize-AM_005fINIT_005fAUTOMAKE-invocation
configure.ac:10: installing './compile'
configure.ac:14: installing './config.guess'
configure.ac:14: installing './config.sub'
configure.ac:5: installing './install-sh'
configure.ac:5: installing './missing'
Makefile.am:6: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
Makefile.am: installing './INSTALL'
Makefile.am: installing './depcomp'
Running autoconf ...
Running ./configure --enable-maintainer-mode ...
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking whether make sets $(MAKE)... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
[... 58 lines cut ...]
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for python... /usr/bin/python
checking for python version... 3.8
checking for python platform... linux
checking for python script directory... ${prefix}/lib/python3.8/site-packages
checking for python extension module directory... ${exec_prefix}/lib/python3.8/site-packages
checking for Python headers using /usr/bin/python-config --includes... ./configure: line 9653: /usr/bin/python-config: No such file or directory
failed, will try another way
  File "<string>", line 1
    import sys; print sys.prefix
                      ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import sys; print sys.exec_prefix
                      ^
SyntaxError: invalid syntax
checking for Python headers in  and ... -I/include/python3.8
checking whether those headers are sufficient... no
configure: error: install python-devel

Pito Salas
Faculty, Computer Science
Brandeis University

pitosalas, 2021-06-16:
Thanks a lot! Looks highly promising, I appreciate it! I hit this error, have you seen it?

checking for Python headers using /usr/bin/python-config --includes... ./configure: line 9653: /usr/bin/python-config: No such file or directory
failed, will try another way
  File "<string>", line 1
    import sys; print sys.prefix
                      ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import sys; print sys.exec_prefix
                      ^
SyntaxError: invalid syntax
checking for Python headers in  and ... -I/include/python3.8
checking whether those headers are sufficient... no
configure: error: install python-devel

-------> Here's the full log. 

[onboard:bullet]~/catkin_ws/python-gudev$ ./autogen.sh

**Warning**: I am going to run `configure' with no arguments.
If you wish to pass any to it, please specify them on the
`./autogen.sh' command line.

processing .
Running libtoolize...
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
Running aclocal -I m4  ...
Running automake --gnu  ...
configure.ac:5: warning: AM_INIT_AUTOMAKE: two- and three-arguments forms are deprecated.  For more info, see:
configure.ac:5: https://www.gnu.org/software/automake/manual/automake.html#Modernize-AM_005fINIT_005fAUTOMAKE-invocation
configure.ac:10: installing './compile'
configure.ac:14: installing './config.guess'
configure.ac:14: installing './config.sub'
configure.ac:5: installing './install-sh'
configure.ac:5: installing './missing'
Makefile.am:6: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
Makefile.am: installing './INSTALL'
Makefile.am: installing './depcomp'
Running autoconf ...
Running ./configure --enable-maintainer-mode ...
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking whether make sets $(MAKE)... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
[... 58 lines cut ...]
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for python... /usr/bin/python
checking for python version... 3.8
checking for python platform... linux
checking for python script directory... ${prefix}/lib/python3.8/site-packages
checking for python extension module directory... ${exec_prefix}/lib/python3.8/site-packages
checking for Python headers using /usr/bin/python-config --includes... ./configure: line 9653: /usr/bin/python-config: No such file or directory
failed, will try another way
  File "<string>", line 1
    import sys; print sys.prefix
                      ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import sys; print sys.exec_prefix
                      ^
SyntaxError: invalid syntax
checking for Python headers in  and ... -I/include/python3.8
checking whether those headers are sufficient... no
configure: error: install python-devel

Pito Salas
Faculty, Computer Science
Brandeis University

mukmalone, 2021-06-19:
Sorry for being slow coming coming back to you.

Not sure if this helps but compared to the steps in the install file I had
to modify 3 other instructions that touched python:

python-gudev ***replaced with gir1.2-gudev-1.0

*** added sudo apt install python-is-python3

sudo easy_install pip ***replaced with sudo apt install python3-pip

pitosalas, 2021-06-19:
Hi Michael

Thanks for the response! I ended up solving it myself by just erasing everything and starting over. So I am all set now. Thanks again.

Pito Salas
Faculty, Computer Science
Brandeis University